bdwilson/KP3S #4: Z offset issue

navivfr, 2022-05-30:
I have this version
- TITAN extruder with metal protection
- Board 1.3 (303).
- BL touch.

Use this firmware :
https://github.com/bdwilson/KP3S/tree/main/releases/2.0.9.3

**ISSUE:** I cannot home the Z.
I can move it up, but never down...

bubbaiOS, 2022-05-30:
Does it work with an older version?

navivfr, 2022-05-31:
Same with the 3 versions
https://github.com/bdwilson/KP3S/tree/main/releases

Doing AUTO HOME (Option)
=> the X an Y doing well.
=> **But the Z rise up** 
=> Then BL touch doing pin IN / OUT several times

And I cannot move the Z axis DOW with the panel control (only UP)

My Board is this one :
[https://www.digital360.solutions/KPS3.jpeg](url)

bdwilson, 2022-06-04:
It works fine for me. https://bubba.d.pr/fyy8Mm

navivfr, 2022-06-05:
Dear Wilson
You used the latest Marlin's version ?
The 'update procedure"  it's just to put the **robin_nano.bin** in the SD card and switch on the printer ?
Or i missed something ?

printCrab, 2022-06-17:
@navivfr You have the GD32F303 RET6 chip mcu

@bdwilson could this be an issue? some are STM32 some are GD32F303.

seems to be an issue with Ender3 pros
[https://github.com/MarlinFirmware/Marlin/issues/23806](url)

icreated4333, 2023-01-18:
@bdwilson, I noticed this issue of z not homing and fixed it, there was a minor error with your code where the z end stop needed to be inverted, I changed it from false to true as seen below. This is from configuration.h.

#define X_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.  //this was initially false but after changing it to true, z-homing works. This should fix the issue.
 
#define I_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define J_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define K_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define U_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define V_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define W_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define I_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define J_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define K_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define U_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define V_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define W_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.

icreated4333, 2023-01-18:
I just realised someone has already posted this solution (@r3xpook) but its here if you need it!