bdwilson/KP3S #8: MCU GD32 F303 compatible

navivfr, 2022-11-28:
Hello,

- I have a MCU GD32 F303 with BL Touch.
- I installed latest your firmware.

But seems to be an issue with Z axis : it doesn't want to go down (event with motion I cannot get it down).

When doing Bed leveling (after setting up Z offset):
•  X/Y axe move to home, 
• But for Z homing , the head move to plate center, Z move up a bit
• probe try to nivel
then nothing...

I missed something ?

BR

navivfr, 2022-11-28:
Dear Wilson,

Thank you for replying...

I guess that is the trick, since I used the BL touch with the KP3S 'normal' firmware - so the BL touch is connected like Kingroon said...

But whet happen to the Z Stop then ? Is not used anymore ?

BR

bdwilson, 2022-11-29:
I had issues getting bltouch to work using their documentation. This was 2 years ago pre-titan. I will look at this the next time I make an update.

navivfr, 2022-11-29:
Hello,

thanks  a lot.. working now (I must read more carefully...)
Still some questions (sorry).
When auto level:
- How to select the matrix number point and range (borders)  ?
- when levelling - the extruder and bed warm up -> how to set those values  (i'm using mostly PETG so bed is 100°, and extruder I don't need to warm it up)
- How to set the approach speed and close level speed ?

With KP3S my robin_nano_cfg.txt look like this :

```
>GRID_MAX_POINTS_X	 5	
>GRID_MAX_POINTS_Y 	5		

>Z_CLEARANCE_DEPLOY_PROBE	2	
>Z_CLEARANCE_BETWEEN_PROBES	10	

>LEFT_PROBE_BED_POSITION 	24	
>RIGHT_PROBE_BED_POSITION 	175
>FRONT_PROBE_BED_POSITION 5
>BACK_PROBE_BED_POSITION 	175

>XY_PROBE_SPEED		9000
>Z_PROBE_SPEED_FAST	1000
>Z_PROBE_SPEED_SLOW 	60
```
BR

navivfr, 2022-12-03:
ok... i recompile all with my parameters...