bdwilson/KP3S #9: Wrong bed size for Marlin 2.1.1

justin-cotarla, 2022-11-30:
It seems the bed size is incorrectly defined for the Marlin 2.1.1 configurations. They are set as:

```
#define X_BED_SIZE 180
#define Y_BED_SIZE 172
```

I believe it should be:
```
#define X_BED_SIZE 180
#define Y_BED_SIZE 180
```

navivfr, 2022-11-30:
there is also an offset of -4mm for Y (I assume is way to make the 'runaway' for filament ?)
So should be like 176 ???

bdwilson, 2022-11-30:
It's so bltouch won't probe on a bed clip for a glass bed.

navivfr, 2022-12-03:
Was also hight wrong (120mm)
On what i understand he use Glass bed with clip, so he loose surface (PEI plate is fantastic BTW)
If you want i made a compilation upon is config with 
- full dimension (180x180x180, but its possible ti wish it to 190 without any problem)
-  7 points bed levelling.
- Temp changement for auto level (i can compile on with you parameters)
-  Black backgroud and some color change

If you interest in it, But I don't know where to push it.

navivfr, 2022-12-03:
Hello, i created a folder with readme;. look like you have to validate ?
you can change the folder name to.
But i will be able to upload the .bin too
BR

bdwilson, 2022-12-03:
You'll have to do a pull request.

navivfr, 2022-12-05:
i dont know how to do this

tildebyte, 2023-01-19:
The weird thing is that the [header file in the repo](https://github.com/bdwilson/KP3S/blob/6018176fc410d797c8650996117976d38cbcc0fb/Marlin-2.1.1/Marlin/Configuration.h#L1698) has 180, but the [header file in the release](https://github.com/bdwilson/KP3S/blob/8d84355710f861bb74633c35365581735ad88a11/releases/2.1.1/Configuration.h#L1698) has 120.

I'm a total newb to this, so I don't know if the running firmware references the header file or not, but it seems like "what is the compiled firmware using" is the real question.