bdwilson/KP3S #13: Non-bltouch 2.1.1

KnownOriginator, 2023-01-11:
Trying to use the 2.1.1 firmware for a my KP3S Pro (I changed the bed size to 210x210, hopefully that's fine), when I try to home the Z axis the gantry just kinda goes up a bit then stops. X homed fine, Y had to be reverse in firmware but that's fine. Z though, that's giving me a fit. I've reverted to the stock firmware from Kingroon for now.

icreated4333, 2023-01-18:
in configuration.h, change the z endstop from false to true. I had the same issue and this fixed it.

#define X_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. // this was initially false

finland777, 2023-09-13:
> in configuration.h, change the z endstop from false to true. I had the same issue and this fixed it.
>
> #define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. #define [...]

Are you able to upload a robin_nano.bin file with the corrections? I'm struggling to learn how to build a new file with the correction.

anthonysedore, 2025-02-04:
[Robin_nano.zip](https://github.com/user-attachments/files/18649743/Robin_nano.zip)